aprsorg/aprs-deviceid #2: Please add tocall APECAN

tkrahn, 2015-03-13:
APE??? was originally allocated for telemetry devices. Therefore I used APECAN as a special subset for my open source / open hardware balloon trackers since they are also kind of telemetry devices.

 "APECAN" : {
"model" : "Pecan Pico APRS Balloon Tracker",
"class" : "tracker",
"vendor" : "Open Source / Open Hardware by KT5TK/DL4MDW and DL7AD"
},

See
https://kt5tk.wordpress.com/
https://github.com/tkrahn/pecanpico5a

Thanks,

Thomas

hessu, 2015-03-13:
Hi,

I'll bounce this to Bob Bruninga over email so that we can it added in tocalls.txt first. That's still the master index for now.

hessu, 2015-03-13:
Bob added it (http://aprs.org/aprs11/tocalls.txt).

The vendor strings would be good to be shorter, compare to the other examples at https://github.com/hessu/aprs-deviceid/blob/master/tocalls.yaml - this will typically be shown on a single line, sometimes on a small mobile device. I'll put KT5TK/DL7AD for the time being?

tkrahn, 2015-03-14:
Dear Heikki,

Thank you very much! Great to have it official and not to conflict with 
anyone else.

Thomas

hessu, 2015-03-14:
You're welcome!

closed 2015-03-14 by hessu