aprsorg/aprs-deviceid #6: Request for new device IDs

DL2MF, 2015-07-07:
Hello Hessu,

I did develop some gateways to APRS during the last few months already registered by Bob in April and now I'd like to request these two new identifiers being added to the tocalls.yaml database enabling them to be identified on aprs.fi.

The first one is for a new APRS WX station gateway supporting the Tinkerforge weather station kit:

tocall:  APR2MF
vendor: Mike, DL2MF
model: MF2wxAPRS Tinkerforge gateway   
class: wx
os: windows

The second is for my WX balloon tracking tool, which is mainly different from other tools by handling radiosonde as an APRS object (not an APRS station like many hams submitting using other tools) and submitting telemetry data to generate temp, wind, height and pressure charts (the last needs already some more bug fixing by myself):

tocall: APB2MF
vendor: Mike, DL2MF
model: MF2APRS tracking tool 
class: software
os: windows

mni tnx fr ur gd wk on aprs community!

73 de Mike, DL2MF

[image]

DL2MF, 2015-07-14:
Hello Heikki,

if there is anything missing to complete my request, please let me know.

tnx es vy 73

Mike, DL2MF

hessu, 2015-11-19:
Thanks for the update, the changes are finally merged. Pull request to directly update the .yaml file is the easiest to merge for me in future.

closed 2015-11-19 by hessu