aprsorg/aprs-deviceid #8: Please add tocall APDNO

DO3SWW, 2016-01-21:
Hello Heikki,

can you add this new software/device to your database:

tocall: APDNO
   vendor: DO3SWW
   model: APRSduino
   class: software
   os: arduino

Bob added it already to his "tocalls.txt".

Thanks
Simon

hessu, 2016-01-25:
Hi,

The assignment is for APDNOx (wildcard for additional version number/model identifier character in the end), not APDNO. Are you going to use APDNO, or APDNO1 or such?

DO3SWW, 2016-01-25:
I wont to use APDNOx. 
x schoud be the current version number.
The first Version of the Software is therefore APDNO1.

hessu, 2016-01-25:
So you will actually use APDNOx, where x is a version number, as documented in tocalls.txt, not APDNOx. Not APDNO as requested in this ticket originally. Ok, I'll add that.

hessu, 2016-01-25:
Also, class: software is for desktop computer software. I'm guessing this is an APRS tracker (something that goes between a GPS and a radio)?

DO3SWW, 2016-01-25:
It's software for a arduino, but when the class software is only for desktop computers, the right class will be tracker.

hessu, 2016-01-25:
Added in ef525f108af315971a7a2b3c4e7264de8829828e. It'll go to aprs.fi when it's next upgraded!

closed 2016-01-25 by hessu