aprsorg/aprs-deviceid #12: Should we add a new class?

LA3QMA, 2016-10-23:
Some radios have APRS, D-Star and/or Fusion.

We already have:
 - class: dstar
   shown: D-Star
   description: D-Star radio

Should there be 2-3 more?

 - class: aprs_dstar
   shown: APRS & D-Star
   description: APRS and D-Star radio

 - class: aprs_fusion
   shown: APRS & Fusion
   description: APRS and Fusion radio

At least radios with both APRS and D-Star should have it's own class.

LA3QMA
Kai Gunter

hessu, 2016-10-23:
That sort of approach would lead to an ever-expanding list of various combinations (aprs_fusion_dmr, aprs_dmr, dmr_dstar ...). I'd rather just put those in as feature flags, one flag per feature, and leave the class to describe form factor (ht, mobile, etc).

Let's come back to this issue next month.

PhirePhly, 2016-11-18:
The Dstar class is for when the APRS traffic source is from Dstar (DPRS), not that the radio supports both APRS and Dstar. Are there Fusion location to APRS gateways? That'd be what warrants a Fusion class, unless we would when want to group all of them under a "gateway" class

hessu, 2016-11-20:
No, the dstar class was for D-Star radios, it's not for the data source. The whole device index is for indexing devices, not for documenting the path the packet might have taken. Now when radios start to do multiple things, the classification needs a little more work; maybe restricting the class to physical form factor (software, hand-held, mobile, tabletop, or such), and then a set of separate feature support identifiers for protocol support if needed.

Q constructs and callsigns after that are for the gateways.

closed 2018-10-25 by hessu