aprsorg/aprs-deviceid #14: Request id for TinyAPRS

shawnchain, 2017-03-18:
- tocall: APTI01/ APTIXX where xx will be the revision numbers
- vendor: BG5HHP(shawn.chain@gmail.com)
- model: TNC/Tracker/Digi
- OS: No OS
- Features: An ATMEGA328p based with KISS TNC/Tracker/Digi features.

The project TinyAPRS:  http://github.com/shawnchain/aprs/

hessu, 2017-03-18:
Hi,

Please read through the instructions in https://github.com/hessu/aprs-deviceid first, and more specifically, please get in touch with Bob first to have it in his tocalls.txt index which is currently the primary source. APTIXX seems to be taken already unfortunately. Also, 'TI' won't work for mic-e due to technical reasons, Bob can assign you something else.

shawnchain, 2017-07-18:
Hi,
Update: WB4APR has assigned the APTAxx for TinyAPRS.
Thanks,

Shawn

PhirePhly, 2017-09-04:
Just an observation, but APTA* was already taken. Bob assigned you a TNCID that was already assigned to K4ATM's tiny track. I don't know how we want to handle this in this database.

tocalls.txt:

      APTAxx  K4ATM's tiny track
      APTAxx  TinyAPRS by BG5HHP

shawnchain, 2017-09-05:
Just confirmed with Bob(WB4APR) that new id APTBxx will be assigned to device "TinyAPRS by BG5HHP"

shawnchain, 2018-02-01:
tocall: APTBxx where xx will be the revision numbers
vendor: BG5HHP(shawn.chain@gmail.com)
model: TinyAPRS TNC
OS: No OS
Features: An ATMEGA328p based with KISS TNC/Tracker/Digi features.

hessu, 2018-10-25:
Thanks for the ticket, and sorry for the slow reply. Added APTB??.

closed 2018-10-25 by hessu

shawnchain, 2018-10-27:
Thank you for the update.