aprsorg/aprs-deviceid #18: easyAPRS

ayasystems, 2017-10-29:
Could you assing me a new device id?

vendor: EA4GKQ ( ea4gkq@ayasystems.com )
model: Tracker
OS: No OS
Features: Two diferent devices:
An ESP8266 + GPS . It sends our position over tcp/ip
An Arduino nano + GPS + DRA818. It will sends beacons over RF

I'm going to public the code when all will be ready and the PCB design. Al free of course.

Regards

ayasystems, 2017-11-04:
You can find easyAPRS details here ->  https://github.com/ayasystems/EASY-APRS

hessu, 2018-10-25:
Hi,
Sorry for the slow reply. Please check the README here, under the title "Adding new devices", and get in touch with Bob Bruninga to have an ID assigned:
https://github.com/hessu/aprs-deviceid

closed 2018-10-25 by hessu