aprsorg/aprs-deviceid #20: Request new entry for APRARX (radiosonde_auto_rx)

darksidelemm, 2018-12-05:
Hi,

Bob has just approved my tocall entry of APRARX, for my open source radiosonde_auto_rx software, which is available here: https://github.com/projecthorus/radiosonde_auto_rx

Vendor: Open Source
Model: radiosonde_auto_rx
class: software
os: Linux/Unix

If this could be added, that would be much appreciated!

Thanks,
Mark VK5QI

hessu, 2019-01-19:
Hi,

If you're into programming, could you consider creating a Pull Request for tocalls.yaml yourself? Don't run the generator, I'll do that after the merge. I'm a bit busy with other stuff in the aprs.fi system, so any help is appreciated.

darksidelemm, 2019-01-19:
About to head off to a conference, but will do when I get a spare minute...

hessu, 2019-01-20:
Thank you very much! I'll deploy this in the next update round of aprs.fi, pretty soon.

closed 2019-01-20 by hessu