aprsorg/aprs-deviceid #21: New APRS device ID add request to tocalls.yaml

F5OPV, 2019-01-09:
Hi Hessu,
Thanks first for your answer on discussion group.
Could you please add the following new device in the aprs.fi database:

   tocall: APSFWX
   vendor: Pierre SAINT-FELIX-CASTETS, F5OPV
   model: Station Météo embarquée
   class: wx
   os: embedded

Thanks in advance Hessu and best 73 de F5OPV.

hessu, 2019-01-19:
Hi,

If you are into programming, could you perhaps create a Pull Request to make the change yourself? I'm a bit busy with some other tasks with the aprs.fi system, so any help is appreciated.

hessu, 2019-01-19:
No need to run the generator or touch the other files, just modify tocalls.yaml alone.

F5OPV, 2019-01-20:
> Thank you for the PR. Merged!
> I'll deploy it on aprs.fi in the next upgrade round, pretty soon.

Thanks to You Hessu.

73 de f5opv

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10