aprsorg/aprs-deviceid #31: New aprs-device

githubmicha, 2019-10-03:
hi,
could you please create a new 'tocall' for arduino-driven (Arduino or ESP8266-Boards) devices.

tocall: apard?
vendor: DL2RXE
model: Arduino/ ESP
class: embedded

thanks in advance
Michael, DL2RXE

hessu, 2019-11-21:
Hi,
This would first have to be added to Bob's master list at http://aprs.org/aprs11/tocalls.txt to prevent overlapping allocations.

githubmicha, 2019-12-10:
How can I add an entry to Bob's list? My mail to Bob (wb4apr@amsat.org) was not answered.

hessu, 2020-02-01:
Bob has added a set of tocalls in January, but oddly not this one in that set. Maybe he just forgot or missed the email.

I'll try to check with him if he's going to be adding these in the future, as soon as he gets out of hospital ( http://lists.aprs.net.au/pipermail/ozaprs/2020-January/014538.html ).

hessu, 2022-02-18:
Hi, as Bob is unfortunately no longer with us, for now I'll be taking in new allocations directly to this database.

The repository is now at https://github.com/aprsorg/aprs-deviceid and this ticket was moved over with it so there's no need to create a new one. There is a new allocation policy document which was discussed with the PR for a few days: https://github.com/aprsorg/aprs-deviceid/blob/master/ALLOCATING.md

Please read through the allocations policy and consider if the request would need to be adjusted to meet it. Thanks!

hessu, 2023-06-05:
Closing ticket due to no activity.

closed 2023-06-05 by hessu