aprsorg/aprs-deviceid #35: `...... X  AP510 KD7LXL    (not message capable spec violation)

lwkoon, 2020-06-10:
Helping the owner of BI7NOR and BG6QBV to raise issue here as they are behind the great wall and he said you somehow miss his msg.

发件人： 阿兵（QBV）
发送时间： 2017-09-04 22:25
收件人： yupopp
抄送： tom; bruninga
主题： 回复： Re: AP510 identifier update
Hello:

I want change this " X" to "_X"
mice:
suffix: "_X"
vendor: AVRTx.cn
model: AP510
class: tracker

tocalls:
tocall: APAV??
vendor: BG6QBV & BI7NOR
model: AVRTx.cn
class: tracker/Igate/Digipeater

Can you help us?

Thanks!

bg6qbv

********************************

wanted to change the suffix from "{SPACE} X"  to "_X"

hessu, 2020-09-26:
Hello,

Please contact Bob Bruninga and have the change done in http://aprs.org/aprs12/mic-e-types.txt first, as that is the master document currently allocating this. For now I'm just tracking changes there, and these documents should be kept in sync.

As for APAV??, only APAVT5 is allocated in http://aprs.org/aprs11/tocalls.txt and not the whle APAV?? prefix.

hessu, 2020-09-26:
Besides, in the mic-e allocation the "_" is all Yaesu radios, I think it is not the best approach to put these devices there at this point.

hessu, 2021-09-18:
Closing, due to no updates.

closed 2021-09-18 by hessu