aprsorg/aprs-deviceid #41: New APRS Device ID

barisdinc, 2020-10-13:
HI,

Bob added our new device to database (tocalls.txt), we will be veru happy if you can add it do device ist as well.
Here is the information ;

tocall: APIZCI
vendor: IZCI APRS Tracker by TA7W/OH2UDS and TA6AEU
model: hymTR IZCI
class: embedded
os: none

hessu, 2020-10-14:
Hi, how about this:

 - tocall: APIZCI
   vendor: TA7W/OH2UDS and TA6AEU
   model: hymTR IZCI Tracker
   class: tracker
   os: embedded

The classes are listed in the beginning of the tocalls.yml document, this is apparently a tracker. The os: field is set to embedded for OS-less devices. I also cleared out the distinction between vendor (maker) and model; are you happy with this model string?

barisdinc, 2020-10-14:
yes, thank you.... I just confused while filling those, but now it is clear
Thanks

Baris
OH2UDS/TA7W

hessu, 2020-10-14:
Thanks, done!

closed 2020-10-14 by hessu

barisdinc, 2020-12-01:
hi,
It's been a month that the tocall files updated but still can't see the DEVICE in aprs.fi ... Do I have to do something else to get APIZCI being decoded right ?

[image]

barisdinc, 2020-12-09:
Hi,

Sorry for bothering you again, but for IZCI Tracker the type is reported as
D-STAR on aprs.fi, what else should I do to get it reported as izci
tracker...

Thanks
Baris
OH2UDS / TA7W

[image: izci.png]
please check the area in *RED BOX*