aprsorg/aprs-deviceid #46: APW9xx

9a9y, 2021-03-24:
Hello Hessu,
Here are my data:

 - tocall: APW9??
   vendor: Mile Strk, 9A9Y
   model: Katarina
   class: wx
   os: emmbeded
   features:
     - messaging
   description: WX station, Tracker, Digipeater, Balloon tracking

Bob say that you make all about this. 
Thank you very much,
Best regards,
Mile, 9A9Y

9a9y, 2021-03-28:
Hello Hessu,
Please, have you time to write my data to your database?
I try to change also all in your new telemetry format as you propose and as aprs.fi work now.

Best regards,
Mile, 9A9Y

hessu, 2021-04-21:
Hello,

Added in e544725eb5951ff6a7d4cb8f763b17d8deea0898 . There can be only one class, so I used "wx" for now as per the model. Let me know if you want to change or remove the class. There is no description field for the tocall - if it would have an awful lot of space for description and "advertisement space", I would expect a lot of edits would come in. :)

closed 2021-04-21 by hessu

9a9y, 2021-04-21:
Hello Hessu,
Thank you very much.
You done all good. Primary that is really wx, so no problem at all from my side.
Greetings from Croatia.
Mile, 9A9Y

9a9y, 2021-04-21:
Hi Hessu, You done all OK. Thank you very much. Best regards, Mile, 9A9Y-------- Izvorna poruka --------Šalje: Heikki Hannikainen ***@***.***>Datum: sri, 21. tra 2021. 23:49Prima: hessu/aprs-deviceid ***@***.***>Kopija: Mile ***@***.***>, Author ***@***.***>Predmet: Re: [hessu/aprs-deviceid] APW9xx (#46)
Hello,
Added in e544725 . There can be only one class, so I used "wx" for now as per the model. Let me know if you want to change or remove the class. There is no description field for the tocall - if it would have an awful lot of space for description and "advertisement space", I would expect a lot of edits would come in. :)