aprsorg/aprs-deviceid #57: New device APATAR

barisdinc, 2021-12-15:
Could you please add the following debve (it is already in Bob’s file)

tocall: apatar
vendor: TA7W/OH2UDS Baris Dinc
model: Arduino ATA-R
class: embedded

Thanks
baris OH2UDS

barisdinc, 2021-12-15:
I will be happy if you can change the class to digipeater, thanks.

hessu, 2022-02-09:
Ok, thanks for the note, adding it now. I hope there are no collisions, as previously there was an APATxx allocation for AnyTone, a Chinese manufacturer which has not been great with allocating identifiers. I've made it like this so that the vendor information does not mix up in the model field:

 - tocall: APATAR
   vendor: TA7W/OH2UDS Baris Dinc and TA6AEU
   model: ATA-R APRS Digipeater
   class: digipeater

closed 2022-02-09 by hessu

barisdinc, 2022-02-10:
Hi,
As you mentioned we are having issues with Anytone, our device is being reported as "Anytone" on aprs.fi details page. 
https://aprs.fi/info/a/TA7W-2
How can we fix this issue? Can it be a solution to move it up before "tocall: APAT??" 

Thanks

hessu, 2022-02-10:
Hi,

The longest match is searched for, order in file does not matter. It'll get updated when aprs.fi fetches a new copy of this database.

73!

commit 1554bc10 (2022-02-09): + APATAR ATA-R APRS Digipeater

closes #57