aprsorg/aprs-deviceid #59: YU1TTN TOCALL allocation

yu1ttn, 2022-02-10:
Hi,

I've tried reaching WB4APR weeks ago, but yesterday I saw the sad news about him going SK.
I'd like to allocate these 8 TOCALLs for my current projects.

tocall: APTNAV
vendor: Filip YU1TTN
model: Tango Navigator
class: tracker
 
tocall: APTNBC
vendor: Filip YU1TTN
model: Beacon station
class: embedded
 
tocall: APTNEX
vendor: Filip YU1TTN
model: Experimental station
class: embedded
 
tocall: APTNIS
vendor: Filip YU1TTN
model: APRSBot
class: app
 
tocall: APTNMT
vendor: Filip YU1TTN
model: Mobile Tracker
class: tracker
 
tocall: APTNND
vendor: Filip YU1TTN
model: APRS Node
class: digi
 
tocall: APTNTL
vendor: Filip YU1TTN
model: Telemetry station
class: embedded
 
tocall: APTNWS
vendor: Filip YU1TTN
model: Weather station
class: wx

73,
Filip YU1TTN

hessu, 2022-02-11:
Hello,

Let's wait for some time until the situation is clarified on APRSSIG, whether this database will be the future master database or not, just to avoid confusion and mismatches.

Are these future or existing projects, and are they software or hardware which are distributed for others to use, or just your personal stations? For example, does "Experimental station" really need its own destination callsign or could you put "Experimental station" in the comment field of the packet to communicate this?

hessu, 2022-02-18:
Hi,

The discussion on APRSSIG was settled, and I can do new allocations now. The repository is now at  https://github.com/aprsorg/aprs-deviceid and this ticket was moved over with it so there's no need to create a new one. There is a new allocation policy document which was discussed with the PR for a few days: https://github.com/aprsorg/aprs-deviceid/blob/master/ALLOCATING.md

Please read through the allocations policy and consider if the request would need to be adjusted. It is unprecedented that someone allocates 8 new identifiers at a time, and it might be difficult over time if everyone did that, so please consider if it is necessary that all of these stations have unique tocalls, and if all of them are different software, and if all of them shall be distributed for others to use. Thank you!

Hessu, OH7LZB

yu1ttn, 2022-02-26:
Hello Hessu,

Thank you for your response!
Is it possible then to allocate APTTNx or APTNGx since I'm planning to
distribute several models (digipeater, weather station/telemetry station,
car tracker and handheld messaging capable tracker)? Last letter would be
used to distinguish different models. For me it's alright if they all go by
general name - Tango Tracker, since I'll put additional description within
status message.

tocall: APTTNx (or APTNGx)
vendor: Filip YU1TTN
model: Tango Tracker
class: tracker

73,
Filip YU1TTN

hessu, 2023-06-05:
Hi,

Sorry for the very long delay.

I've allocated APTNG? as suggested, APTT was already allocated for Byonics.

closed 2023-06-05 by hessu

yu1ttn, 2023-09-04:
Hi Heikki,

Thank you for the update and allocation!
I'll make sure to switch all devices to the new TOCALL.

73,
Filip YU1TTN/SU9FP

commit 69a3d71d (2023-06-05): + APTNG? Tango Tracker, YU1TTN

closes #59