aprsorg/aprs-deviceid #62: Proposed TOCALL allocation

W0CHP, 2022-05-09:
I am proposing that the popular [WeeWX weather software](https://github.com/weewx/weewx) has a TOCALL allocation:

```yaml
- tocall: APWX??
  vendor: Tom Keffer and Matthew Wall
  model: WeeWX Weather Software
  class: software
  os: Linux/Unix
```

The goal is to replace the existing TOCALL packet preamble in WeeWX from `APRS` to `APWX??`:
https://github.com/weewx/weewx/blob/7024f046cffcff145af4e195cefc18f9c08cd3fb/bin/weewx/restx.py#L1231

hessu, 2023-06-05:
Hi,

Sorry for the terribly long delay. Allocated APWX?? to WeeWX. I run it too at one location.

Can you do the PR for WeeWX to use the new prefix?

closed 2023-06-05 by hessu

hessu, 2023-06-05:
Sorry, decided to reallocate from APWX?? to APWEE? to make it a 5-letter prefix. We only have 676 4-letter prefixes in the AP?? space available.

W0CHP, 2023-06-05:
Sure can!  I’ve contributed plenty to the project, Thanks!

W0CHP, 2023-06-05:
No worries at all, that works great. I’ll work with Tom and Matt on the weeex change/PR

W0CHP, 2023-06-07:
> 
> Can you do the PR for WeeWX to use the new prefix?

FYI: Just submitted the WeeWX PR:
https://github.com/weewx/weewx/pull/870

:-)

commit 6a45505f (2023-06-05): + APWX?? for WeeWX weather station software

closes #62
commit 6f67c185 (2023-06-05): + APWEE? for WeeWX weather station software

closes #62