aprsorg/aprs-deviceid #66: New APRS Device ID

erstec, 2022-10-15:
Please add new TOCALL for my new embedded APRS Tracker / IGate / Digi Open Source Project - APRS-ESP
 - tocall: APES??
   vendor: LY3PH
   os: embedded

It's in hw/fw testing / pre-release stage now.
Last '??' digits will be used for versioning/hardware/features separation

73, LY3PH / Ernest

erstec, 2022-11-12:
Project assembled, tested and released, still with experimental destination.

hessu, 2023-06-05:
Hi,

Sorry for the very long delay. I have assigned you with APESP?, I hope that's alright. We only have 676 4-letter prefixes in the AP?? space available, with 5-letter prefixes there's more space to grow.

erstec, 2023-06-06:
> Hi,
> 
> Sorry for the very long delay. I have assigned you with APESP?, I hope that's alright. We only have 676 4-letter prefixes in the AP?? space available, with 5-letter prefixes there's more space to grow.

Thank you!
Due to mentioned limitations APESP? is ok of course!

73!

oh2th, 2024-03-17:
In May 2020 these two were allocated as:
APESPG = ESP SmartBeacon APRS-IS Client
APESPW = ESP Weather Station APRS-IS Client

Could these be restored as they were?

hessu, 2024-04-05:
Hi,

Unfortunately I have missed the APESPG and APESPW allocations to OH2TH by Bob in 2020, and then subsequently made an overlapping allocation. This needs to be resolved by reallocating.

Here's the number of stations seen in 2024 (according to my database on aprs.fi) using different `APESP?` tocalls:
```
APESPW   2
APESPG   45
APESP1   2
```

Do you have any visibility if these are running the software from @oh2th or from @erstec ?  Have you released code using these tocalls?

erstec, 2024-04-05:
@hessu 
APESP1 is what mine project using, FW is deployed and running on several (at least I know) nodes.
We can split interval, like APESPx, where x is digit are more than enough for me.

hessu, 2024-04-05:
If you're currently using APESP1 only, the simple approach would be to make an allocation of APESP1 to you to resolve the overlap. Would that be alright for you?

erstec, 2024-04-06:
Yes, it's ok, Thank you

oh2th, 2024-04-15:
Sounds like aplan

hessu, 2024-04-29:
Hi,

I have fixed the allocations as agreed above.

commit 69b6c2d3 (2024-04-29): + APESPG APESPW, fix APESP1 overlap

APESPG and APESPW were accidentally omitted previously, and APESP? had
been allocated as an overlapping allocation. As agreed on ticket #66, narrow
down the newer allocation to APESP1 and put APESPG & APESPW back.

closes: #144
closes: #66
closed 2024-04-29 by hessu