aprsorg/aprs-deviceid #76: new  TOCALL Request

richonguzman, 2023-06-05:
Hi, Heikki

(Thanks for processing APLR0? as a Tracker!)

As you message made it clear I should ask now as follows:

- tocall: APLR10
   vendor: Ricardo, CD2RXU
   model: ESP32 LoRa iGate
   class: igate
   os: embedded
   contact: richonguzman@gmail.com

- tocall: APLR13
   vendor: Ricardo, CD2RXU
   model: ESP32 LoRa WX Station
   class: wx station
   os: embedded
   contact: richonguzman@gmail.com

Hope this is what you ment with the Class selection when closing the issue #74 

Ricardo CD2RXU

richonguzman, 2023-06-06:
Heikki,

tocall: APLRTX
model: ESP32 LoRa Tracker 
....and all its variations (Tracker w/ GPS beacons w/ or w/o WX or just a plain WX Station) ... being X -->0-9

tocall: APLRGX
model: ESP32 LoRa iGate 
....and all its variations (iGate, DigiRepeater, iGate over 4G/LTE (w/ or w/o WX info))... being X --> 0-9

Let me know what you think.

hessu, 2023-06-06:
Hi,

Ok. The assignment with a selectable character be with a wildcard in the end, APLRT? and APLRG? where ? may be any number or upper-case letter, [0-9A-Z]. X would be "X", not "any character".

```
 - tocall: APLRG?
   vendor: Ricardo, CD2RXU
   model: ESP32 LoRa iGate
   class: igate
   os: embedded
   contact: richonguzman@gmail.com

 - tocall: APLRT?
   vendor: Ricardo, CD2RXU
   model: ESP32 LoRa Tracker
   class: tracker
   os: embedded
   contact: richonguzman@gmail.com
```

I have made the allocation now as above.

commit c4830823 (2023-06-06): + APLRG? and APLRT? for CD2RXU ESP32 LoRa

closes #76