aprsorg/aprs-deviceid #80: New APRS Identifier

kkaiser1952, 2023-06-07:
tocall: NCM123
Vendor: Keith Kaiser, WA0TJT
Model:
Class: software
Os: browser
Contact: wa0tjt@gmail.com
Features: Net Control Manager is used by over 600 ARES/RACES/RedCross, etc. emergency communicator groups and ham clubs to coordinate their net activities including weekly social nets.
  -- messaging the APRS text messages will include include location information in several formats, lat/lng, W3W, crossroads, additionally an object name and description, possibly more. The quantity of use is expected to be quite low. The exact layout is under development at this time.

hessu, 2023-06-07:
Hi,

The features field is not a free-text field, it is only required for some technical purposes, see https://github.com/aprsorg/aprs-deviceid/blob/main/ALLOCATING.md .

The name of the application goes in the Model field. The APRS tocalls all start with 'AP' according to the spec. So I suppose the following should be right?

```
- tocall: APNCM
  vendor: Keith Kaiser, WA0TJT
  model: Net Control Manager
  class: software
  os: browser
  contact: wa0tjt@gmail.com
```

kkaiser1952, 2023-06-07:
I apologize Heikki I misunderstood the email that came out today. All of this is correct, would you prefer that I resubmit it or can you take care of that? I’m a happy camper either way.

Thanks...

hessu, 2023-06-12:
Hi,

Ok, thanks, I'll create the allocation with the above details. No need to resubmit.

commit 6c4e6804 (2023-06-12): + APNCM for WA0TJT Net Control Manager

closes #80
PR #81 (2023-06-12): + APNCM for WA0TJT Net Control Manager
closes #80
closed 2023-06-12 by hessu