aprsorg/aprs-deviceid #82: New vendor/devices

ure, 2023-06-25:
Hi there,

We are a new vendor just started launching HAM related equipment and kits, after working years developing soft and hardware for 3th parties we want to provide back to the community.

We are currently testing several trackers and igate's, hardware is finalized ... software for the WX is being developed.
There is also a dual band open source HT and mobile rig on it's way. 
We are also working on a solar powered Lora 433 to 2M crossband digipeater.

https://github.com/orgs/Guru-RF/repositories

We propose following ... all feedback/insight is welcome !!

- tocall: APRFG??
   vendor: RF.Guru
   model: Embedded RP2040 APRS devices
   os: embedded
   contact: [info@rf.guru](mailto:info@rf.guru)

 - tocall: APRFGT?
   vendor: RF.Guru
   model: RP2040 LoRa APRS Tracker
   class: tracker
   os: embedded
   contact: info@rf.guru

- tocall: APRFGW?
   vendor: RF.Guru
   model: RP2040 LoRa APRS Weather Station
   class: wx
   os: embedded
   contact: [info@rf.guru](mailto:info@rf.guru)

- tocall: APRFGI?
   vendor: RF.Guru
   model: RP2040 LoRa APRS iGate
   class: igate
   os: embedded
   contact: [info@rf.guru](mailto:info@rf.guru)

- tocall: APRFGD?
   vendor: RF.Guru
   model: RP2040 LoRa APRS Digipeater
   class: digi
   os: embedded
   contact: [info@rf.guru](mailto:info@rf.guru)

- tocall: APRFGM?
   vendor: RF.Guru
   model: RP2040 LoRa APRS Digipeater
   class: digi
   os: rig
   contact: [info@rf.guru](mailto:info@rf.guru)

- tocall: APRFGP?
   vendor: RF.Guru
   model: RP2040 LoRa APRS Digipeater
   class: ht
   os: embedded
   contact: [info@rf.guru](mailto:info@rf.guru)

hessu, 2023-06-26:
Hi,

Mostly the vendor and model are shown, so I suppose a single entry for the "RP2040 LoRa APRS Digipeater" model would be sufficient? A single model with three different classes would be weird?

The destination callsign only has 6 characters in it, so the vendor allocation would be APRFG? with the last character for different models. None of the models would have wildcards after them.

Would this be sufficient?

ure, 2023-06-26:
Yes that's perfect ;) Many thank !

hessu, 2023-07-07:
Hi,

The allocations are done. I also added an APRFG? vendor allocation without a model.

commit a2b0a57c (2023-07-07): + APRFG? for RF.Guru devices

closes #82
commit 8086a00f (2023-07-07): + APRFG? for RF.Guru devices

closes #82
closed 2023-07-07 by hessu