aprsorg/aprs-deviceid #86: Confused about ordering / preference on overlapping tocalls

xssfox, 2023-07-19:
The README.md suggests looking for the longest match however there are cases where you can have a tie based on how many characters match (see example below). I'm guessing that the idea is that we should favour the longest matching prefix first? so `APC???` would be the APRS device ID. Some clarification around this would be good.

What does APCAD match?
```yaml
 - tocall: APC???
   vendor: Rob Wittner, KZ5RW
   model: APRS/CE
   class: app
 - tocall: APnnnD
   vendor: Painter Engineering
   model: uSmartDigi D-Gate
   class: dstar
```

hessu, 2023-07-19:
Hi! APCAD would match APC???, as the 'n' wildcard should only match numeric (0-9) characters. The spec is not spelling that out, I'll fix that.

These are both old assignments and there may be some ambiguity in there, and we should avoid these in the future.

APRS/CE is probably not that frequently met (Windows CE...), I'd guess.

xssfox, 2023-07-19:
That makes sense, thanks!

closed 2023-07-20 by hessu