aprsorg/aprs-deviceid #91: Identifying radio type for APRS device ids

borjam, 2023-08-04:
Hello,

Now that APRS-LoRa is growing but it hasn't grown too much, I think it would be a good idea to consider some way to identify devices based on the radio type used. VHF "analog" TNC, LoRa, maybe others in the future.

I know it's a far reaching change, just suggesting it should be considered.

hessu, 2023-08-04:
A lot of the software can be used with any type of radio interface, so I'm not sure about this.

I'd like to perhaps avoid adding a lot of additional details to the entries. When more data is added, there will be more changes to manage.

borjam, 2023-08-04:
I know it's not a trivial question and it would need careful consideration! But maybe in the future more radio interfaces will appear. Fow now LoRa is offering promising results, and there are even products available in the market. 

So, just seeding an idea!

hessu, 2023-08-13:
Yep, thanks.

Not starting to do this now, for the following reasons:
  * Many devices and software can use various different protocols and modems, not just one. There are already a lot of different methods used (APRS, DMR, various HF modes).
  * If we'd tag software X (running on a computer or mobile phone) as capable of doing APRS, LoRa and DPRS in this database, it would not communicate which protocol is being used at the moment. I presume one of your goals would be to filter stations based on the protocol, but this criteria would not work well for that when multiprotocol devices are out there.
  * Maybe the igates should be tagging themselves somehow, signaling that stations which were received by this igate callsign-SSID were on protocol Y.
  * Adding more details to the database will increase the amount of changes and updates, increasing the effort required to maintain it.

closed 2023-08-13 by hessu