aprsorg/aprs-deviceid #93: New Experimental APRS-Digipeater Board request ID

LAMK92, 2023-08-14:
Hi dear all

We are building an experimental APRS-Payload on VHF Band for a CubeSat digipeater as part of a competition held by Kyutech that is called BIRDSX, so we would like to have an APZAEP for TOCALL and a future for our next Cubesat GuaraniSat-2 with the same mission.

Link of the APRS payload competition: https://birds-x.birds-project.com/satellite-documents/

Could you please allocate it.

tocall: APZAEP
vendor: Paraguay Space Agency (AEP)
model: EIRUAPRSDIGIS&FV1
class: embedded

Thanks in advance
Luis, ZP6LMK

hessu, 2023-09-07:
Hi,

The idea of the APZ prefix for experimental use is that temporary experimental use can be done there without specifically allocating a permanent identifier.

If you are going to actually release software for others to use, or launch a satellite with this software, how about if we allocate **APAEP1**  for this firmware? If it's going on an actual spacecraft (I'm not entirely sure based on your description), it should probably be **class: satellite** .

LAMK92, 2023-09-08:
Hi 
Thanks for the suggestion dear colleague.
The idea is to make an PCB APRS DIGI PAYLOAD for a CUBESAT. 
So I will take your suggestions as follow:

APRS TNC features:
tocall: APAEP1
vendor: Paraguay Space Agency (AEP)
model: EIRUAPRSDIGIS&FV1
class: embedded

So please register.

Thanks in advance

Luis, ZP6LMK

hessu, 2023-10-16:
Hi,

"EIRUAPRSDIGIS&FV1" is not the most beautiful model name I've seen. We can have space characters in the name too.

But OK, registering now with that spec. Let me know in this ticket if you wish to update still.

commit 33cc6488 (2023-10-16): + APAEP1 for Paraguay Space Agency satellite

fixes #93
closed 2023-10-16 by hessu