aprsorg/aprs-deviceid #96: OpenGD77 VHF/UHF radio firmware now has APRS Tx support

rogerclarkmelbourne, 2023-08-29:
Hi 

The OpenGD77 firmware now has APRS Tx in all supported radios.  This includes Radioddity GD-77, Baofeng DM-1801, Baofeng DM-1801A, Baofeng RD-5R, TYT MD-9600 , Retevis RT-90, TYT MD-UV380, , TYT MD-UV390, Retevis RT-3S and Baofeng DM-1701.

Source code for 'stable' is released under a modified BDS 4 clause license which prohits commerical use or commerial derivation.

See https://www.opengd77.com 
Our User Guide is here  https://github.com/LibreDMR/OpenGD77_UserGuide

Note the firmware is always evolving and the User Guide shows the option for people to enter their own "tocall" but this feature is being removed in the next beta and APOG77 used instead 

In the future the firmware may support APRS Rx, but currently its Tx only.

Please can you allocate

tocall: APOG77

Thanks

Roger  VK3KYY/G4KYF

hessu, 2023-09-07:
Hi,

Excellent news!

If you're making different builds for the different radios, or the firmware otherwise knows which radio it is on, we could also allocate you with an APOG7? prefix where the last letter (single-letter wildcard) would indicate the radio model. How about that?  Then the per-radio classes can point to ht/rig as necessary, and per-model metrics can be generated.  This would be consistent with other vendors where different radio models get their own tocalls.

I'm allocating the APOG7? prefix (vendor: OpenGD77, model: OpenGD77) just to get the allocation done, this will catch APOG77 as well. Per-radio tocalls (APOG7A, APOG7B, APOG7C ...) can then be allocated later.

rogerclarkmelbourne, 2023-09-07:
There are separate builds

i.e the project started 5 years ago for only the Radioddity GD77 radio, and has expanded now to the following radios, which each have different firmware builds

Radioddity GD-77 (also sold as TYT MD-760)  - No longer  in production
Baofeng DM-1801- No longer  in production
Baofeng DM-1801A - No longer  in production
Baofeng RD-5R - No longer  in production
TYT-MD9600 (also sold as Retevis RT-90) - Still in active production
TYT MD-UV380 (also sold as Retevis RT-3S) - Still in active production
Baofeng DM-1701 - Still in active production

We hope to add the TYT-MD2017 to this list in the next 6 months, but porting to new radios can take several years.

For a while we did call each of these firmware's by different names e.g. OpenGD77, OpenMD9600, OpenMDUV380 etc etc

But because the operation / UI etc is  99% the same across all hardware platforms, users found this confusion, and we now just call it OpenGD77 for all radios even if its not a Radioddity GD-77. In fact the Radioddity GD-77 is no longer in production, mainly due to ongoing chip shortages

commit f541b5cc (2023-09-07): + APOG7? for OpenGD77 project

closes #96
closed 2023-09-07 by hessu