aprsorg/aprs-deviceid #98: LoRa APRS Meteo by OK2DDS

ddaqua6, 2023-09-03:
Hello,

I would like to request allocation of ALGMx dest call to my project of [LoRa_APRS_Meteo](https://github.com/ddaqua6/LoRa_APRS_Meteo), which is a combination of LoRa igate, digipeater and WX station. There are currently some stations in the world working with this firmware, which uses this prefix.

To sum up my request

- tocall: APLGM?
- vendor: David, OK2DDS
- model: LoRa APRS Meteo
- contact: david@ok2dds.cz

73 and all the best!
David, OK2DDS

hessu, 2023-09-07:
Hi,

APLG?? is already allocated to OE5BPA, so APLGM? cannot be allocated for you, as it would overlap a recently created allocation.

How about `tocall: APLDD?` instead?

ddaqua6, 2023-09-10:
Hello,

then we could change it so we can better distinguish between the two parts of the device. As I mentioned, it's an WX station combined with igate/digipeater. So let's alocate APLDM? to "LoRa Meteostation" (WX) and APLDI? to "LoRa IGate/Digipeater" (digi)

hessu, 2023-10-16:
Hi,

Ok, I'm allocating APLDM? and APLDI? .

commit ff0181ac (2023-10-16): + APLDI? APLDM? to David, OK2DDS for LoRa wx and igate/digi

closes #98
closed 2023-10-16 by hessu