aprsorg/aprs-deviceid #117: ToCall Request: DStarGetway

F4FXL, 2023-12-23:
tocall: APD5T?
vendor: Geoffrey, F4FXL
model: Open Source DStarGateway
class: dstar
os: Multiple
contact: [f4fxl@dstargateway.digital](f4fxl@dstargateway.digital)
features: Advertise DStar Repeaters to APRS-IS, Forwards APRS messages from/to DStar

[https://github.com/F4FXL/DStarGateway](https://github.com/F4FXL/DStarGateway)

F4FXL, 2024-02-05:
Hi, any updates on this?

hessu, 2024-02-09:
Hi!

Sorry for the delay. I've made the allocation now for APD5T?.

hessu, 2024-02-09:
The features field is for predefined flags, not free text, so I've left it out.

closed 2024-02-09 by hessu

F4FXL, 2024-02-09:
> The features field is for predefined flags, not free text, so I've left it out.

Thanks for the pull.
Did not know that, what are the flags?

hessu, 2024-02-10:
The feature flags are listed in README.md, there are just two:
https://github.com/aprsorg/aprs-deviceid/blob/main/README.md#contents

PR #129 (2024-02-09): 6 updates
closes #117 #121 #122 #124 #126 #127
commit fa748ca9 (2024-02-09): + APD5T? for Open Source DStarGateway, F4FXL

closes #117