aprsorg/aprs-deviceid #118: Yaesu FTM-500D

wb2osz, 2023-12-24:
The discussion here  https://groups.io/g/APRS/topic/101247433#1944  provides evidence that "_4" is the suffix for Yaesu FTM-500D.

73,
John WB2OSZ

hessu, 2024-02-09:
Hi,

Yes, it is existent, but the vendor has not made any communication to me or anyone else. It'd be good if they would establish some sort of communication channel to avoid overlapping allocations, and so that we could get in touch with them when bugs occur.

wb2osz, 2024-02-09:
The "tocall" value is also missing.
Based on the pattern seen,

  - tocall: APY300
   vendor: Yaesu
   model: FTM-300D
   class: rig
   
 - tocall: APY400
   vendor: Yaesu
   model: FTM-400
   class: rig

The FTM-500D would probably be APY500 but I have no evidence for this yet.

I sent a note to Yaesu asking them to confirm.

73,
John WB2OSZ

hessu, 2024-02-14:
Update: Yaesu has responded and indicated they will be coordinating this now. Stay tuned for updates.

hessu, 2024-02-16:
YAESU Customer Relations has provided the details over email, and also promised to get in touch when new models are coming out. Handled this in ticket #134 and PR #135 .

closed 2024-02-16 by hessu