aprsorg/aprs-deviceid #119: Yaesu FTM-200D

wb2osz, 2023-12-24:
I noticed a local station, with MIC-E suffix "_2" showing up as unknown vendor/model.
I wrote to the station operator and he said it was a Yaesu FTM-200D.

73,
John WB2OSZ

gsm-com, 2023-12-28:
It seems to be that Hessu only implements deviceid from Yaesu when he gets the official confirmation from them. My request from October was closed with that reason. :-(

na7q, 2024-01-09:
> It seems to be that Hessu only implements deviceid from Yaesu when he gets the official confirmation from them. My request from October was closed with that reason. :-(

That's pretty ridiculous honestly. It's well known knowledge at this point.

wb2osz, 2024-02-09:
The "tocall" value is also missing.
Based on the pattern seen,

  - tocall: APY300
   vendor: Yaesu
   model: FTM-300D
   class: rig
   
 - tocall: APY400
   vendor: Yaesu
   model: FTM-400
   class: rig

The FTM-200D would probably be APY200 but I have no evidence for this yet.

I sent a note to Yaesu asking them to confirm.

73,
John WB2OSZ

hessu, 2024-02-14:
I sent in messages to Yaesu through three channels now, and they responded. They'll be coordinating this now, updates will be coming.

closed 2024-02-14 by hessu

hessu, 2024-02-16:
Yaesu customer relations sent in their new identifiers, and this was processed in #134. They promised to send in future updates as well.