aprsorg/aprs-deviceid #122: Add APNV2? and update APNV0?

sq8vps, 2024-01-21:
Hi, my requests are:
1.  Add new device ID:
```
   tocall: APNV2?
   vendor: SQ8L
   model: VP-Tracker
   class: tracker
```
2. Add new Mic-E ID:
```
   suffix: "$2"
   vendor: SQ8L
   model: VP-Tracker
   class: tracker
```
3. Update APNV0?:
```
   tocall: APNV0?
   vendor: SQ8L
   model: VP-Digi
   os: embedded
   class: digi
```
Thanks!

hessu, 2024-02-09:
Hi,

1. I've allocated APNV2? as requested,
2. '$' is not in the list of planned codes in http://www.aprs.org/aprs12/mic-e-types.txt so just to play safe, I have allocated a suffix of ":2" instead. I hope that's OK.
3. Updated APNV0? as requested.

closed 2024-02-09 by hessu

sq8vps, 2024-02-09:
Haven't seen this file, sorry. Thank you for the changes!

commit 3ef4f441 (2024-02-09): + APNV2? for VP-Tracker, updated APNV0?

closes #122