aprsorg/aprs-deviceid #125: APRS device by IW1CGW

iw1cgw, 2024-02-07:
I would like to request the APLF?? dest call for my projects of APRS: igate, digipeaters, WX stations, telemetry devices.
Currently there are some stations in the world working with my HAM solutions available on GitHub or: https://iw1cgw.wordpress.com/.

To sum up my request

 - tocall: APLFI?
   vendor: Giovanni, IW1CGW
   model: LoRa IGate/Digipeater/Telemetry
   class: digi
   contact: iw1cgw@libero.it	

 - tocall: APLFM?
   vendor: Giovanni, IW1CGW
   model: LoRa Meteostation
   class: wx
   contact: iw1cgw@libero.it

 - tocall: APLFA?
   vendor: Giovanni, IW1CGW
   model:  Meteo
   class: wx
   contact: iw1cgw@libero.it	

 - tocall: APLFB?
   vendor: Giovanni, IW1CGW
   model:  Telemetry
   class: 
   contact: iw1cgw@libero.it

73 and all the best!
Giovanni, IW1CGW

hessu, 2024-02-09:
Hi,

Just to double check, are the two different "meteo" devices different, or is it an accident?

iw1cgw, 2024-02-09:
Hi,
yes, they are different devices, I have a class of devices in LoRa and another in old AX25 at 1200 baud. I wanted to distinguish them - if possible -

hessu, 2024-02-09:
Hi,

Ok. LoRa devices have been recently allocated in the APL* prefix, and people have been using that prefix to filter & display LoRa stations only. So the AX.25 devices should perhaps go to some other prefix

iw1cgw, 2024-02-09:
You are right ! what I proposed is not good for AX25 devices, can you choose You as an alternative or do I open another ticket just for AX25 devices? - For LoRa I think it could be fine, right?

iw1cgw, 2024-02-09:
APHR Class for AX25 ? 
-----------------
tocall: APHRM?
vendor: Giovanni, IW1CGW
model: Meteo
class: wx
contact: [iw1cgw@libero.it](mailto:iw1cgw@libero.it)

tocall: APHRT?
vendor: Giovanni, IW1CGW
model: Telemetry
class:
contact: [iw1cgw@libero.it](mailto:iw1cgw@libero.it)

hessu, 2024-02-09:
Hi,

Ok, thanks. I have allocated APLHI? APLHM? APHRM? APHRT? in ad79d6838062cbd6001f6e63c9fec719929c0247 .

closed 2024-02-09 by hessu

iw1cgw, 2024-02-09:
thanks for your patience ! -- 73 iw1cgw Giovanni @ North West Italy

commit ad79d683 (2024-02-09): + APLHI? APLHM? APHRM? APHRT? for Giovanni, IW1CGW

closes #125
PR #130 (2024-02-09): + APLHI? APLHM? APHRM? APHRT? for Giovanni, IW1CGW
closes #125