aprsorg/aprs-deviceid #132: Request for new tocall

HariHend1973, 2024-02-12:
tocall: APTHUR
model : APRSThursday weekly event mapbot daemon
contact : harihend1973@gmail.com
vendor : YD0BCX
os : linux/unix
features : messaging

Thanks in Advance
hari/YD0BCX

ldeffenb, 2024-02-12:
Will the mapbot daemon actually be generating APRS packets?   If not, then it does not need an assigned ToCall.

HariHend1973, 2024-02-12:
I planned to make the bot send its position to aprs.fi with tactical callsign and replies to stations who query to it..
Right now it just sending packets(keepalive and ack) to ANSRVR and APRSPH with dstcall APZTH1.

I'm looking for any suggestions.
Thanks

ldeffenb, 2024-02-12:
Very good!  I was just curious, and in fact, this is perfect.   If I had thought about it, I would have realized that you must be sending packets to keepalive with ANSRVR.   Duh on me!

HariHend1973, 2024-02-12:
Infact the bot will replied with ack to stations who send messages(with kessage id) directly to it.

I've just thought that would be nice if the bot has an identity for itself  😊

Thanks

hessu, 2024-02-14:
Hi,

Thanks, I'm making the allocation now. Added `class: service`, as it's mainly a service you run, not software for end users to run.

closed 2024-02-14 by hessu

HariHend1973, 2024-02-14:
> Hi,
> 
> Thanks, I'm making the allocation now. Added `class: service`, as it's mainly a service you run, not software for end users to run.

thank you
regards

commit 65b8afcf (2024-02-14): + APTHUR, APRSThursday weekly event mapbot daemon

closes #132
PR #133 (2024-02-14): + APTHUR, APRSThursday weekly event mapbot daemon
closes #132