aprsorg/aprs-deviceid #145: Lora Aprs i-Gate

ta5lvc, 2024-03-26:
tocall: APTIF
vendor: TA5Y
model: TIF LORA APRS I-GATE
class: i-gate

hessu, 2024-04-05:
Hi,

Recently all allocations for LoRA devices have been made in the APL* prefix. Would you be alright with the allocation of APLIFT?

APLTIF is within another existing allocation already.

ta5lvc, 2024-04-06:
It can be APZTIF, APTIF or APYTIF. If there is no allocation regarding
these. Thanks

Heikki Hannikainen ***@***.***>, 5 Nis 2024 Cum, 22:24
tarihinde şunu yazdı:

hessu, 2024-04-29:
Hi,

As I mentioned before: All recent LoRa allocations have been with an **APL** prefix, so I'd prefer to allocate you a tocall which starts with APL, too. APLTIF is already taken, so I'll allocate you the prefix APLIF? where you can use the last wildcard letter/number for variants.

I hope this is alright.

closed 2024-04-29 by hessu

ta5lvc, 2024-04-30:
> completed via 0007901

Ok.

Heikki Hannikainen ***@***.***>, 30 Nis 2024 Sal, 00:53
tarihinde şunu yazdı:

commit 0007901e (2024-04-29): + APLIF? for TA5Y TIF LORA APRS I-GATE

closes #145