aprsorg/aprs-deviceid #147: N1AF LoRa I-gate and tracker

n1af-ham, 2024-04-06:
Greetings!  I'm interested in registering 2 LoRa-APRS tocall's:

 - tocall: APLZY?
   vendor: N1AF
   model: LoRa-APRS Tracker
   class: tracker
   os: embedded
   contact: lora-aprs@n1af.org

 - tocall: APLZZ?
   vendor: N1AF
   model: LoRa-APRS iGate
   class: igate
   os: embedded
   contact: lora-aprs@n1af.org
   
   I'm flexible on the actually assigned tocall's, just included some possible 5-char prefixes that aren't allocated to anything yet.

hessu, 2024-04-29:
Hi,

Just to confirm: Are these actual different devices and different software, or just one software switching between igate / digipeater / tracker mode?

n1af-ham, 2024-04-30:
Hi Hessu,

All 3 are different independent software projects.

-Ryan (N1AF)

hessu, 2024-06-04:
Hi,

For some reason all of the LoRa allocation requests have recently requested 3 to 5 tocalls. This will increase the size of the database and require more maintenance work, so I'm allocating a single prefix for each applicant for now. The last letter can be used by the author to identify variants. I hope this is OK.

```
 - tocall: APLZX?
   vendor: N1AF
   model: LoRa-APRS
   os: embedded
   contact: lora-aprs@n1af.org
```

PR #154 (2024-06-04): 4 allocations
closes #147
closes #149
closes #150
closes #151
closed 2024-06-04 by hessu