aprsorg/aprs-deviceid #156: APRRES - to be released public project

xssfox, 2024-06-26:
```yml
 - tocall: APRRES
   vendor: xssfox
   model: APRS-RepeaterRescue
   class: network
   os: embedded
   contact: repeater-rescue@michaela.lgbt
   features:
      - messaging
```

ldeffenb, 2024-06-26:
Is this related at all to the "Rescue Repeater" at https://www.beagleboard.org/projects/extending-communications-drone-based-radio-repeater

Extending Communications – Drone-based Radio Repeater
The "Rescue Repeater" is a drone-based radio repeater solution to allow communications to be rapidly restored or extended over the horizon.

xssfox, 2024-06-26:
Not related

hessu, 2024-06-26:
Hi,

There's no "network" class defined. I'm guessing this is a service or bot of some sort which sends and/or receives APRS messages? So I'd put it in the "service" class, is that alright?

xssfox, 2024-06-26:
Service works for me. I got network from here

https://github.com/aprsorg/aprs-deviceid/blob/main/ALLOCATING.md#applicationdevice-classes

hessu, 2024-06-28:
Ah right, they were missing in tocalls.yaml while they were defined in ALLOCATING.md. Fixed in #158 .

I'll make the allocation with "network" class which is for hardware appliances.

commit a3a04422 (2024-06-28): + APRRES for xssfox: APRS-RepeaterRescue

closes #156
closed 2024-06-28 by hessu