aprsorg/aprs-deviceid #157: New APRS TOCALL Request: APLDAG EA2CQ APRS & LoRa Spotting Service

EA2CW, 2024-06-27:
Hello, 
We would want to request the TOCALL "APLDAG" for "DAGA", a spotting service from talkies to SOTA (Summits On The Air) 
It is at final stages of development by Iñigo EA2CQ and myself, Mikel EA2CW. 
The data:

tocall: APLDAG
vendor: Inigo, EA2CQ
model: DAGA LoRa/APRS SOTA spotting
class: service
os:Linux
service: APRS message bot
contact: ea2cq@irratia.org
features: messaging

Thank you & 73
Mikel EA2CW | AE2CW

hessu, 2024-06-28:
Hi!

Thanks, I'm making the allocation as follows. Removed the "service" field because there is no such field, and removed the "os" field as it's a service which nobody else will be running; it doesn't make much sense to announce the OS it'll be running on.

```
 - tocall: APLDAG
   vendor: Inigo, EA2CQ
   model: DAGA LoRa/APRS SOTA spotting
   class: service
   contact: ea2cq@irratia.org
   features:
     - messaging

```

closed 2024-06-28 by hessu

EA2CW, 2024-06-28:
Thank you, OM!

commit 0eb9d331 (2024-06-28): + APLDAG for EA2CQ DAGA LoRa/APRS SOTA spotting

closes #157