aprsorg/aprs-deviceid #161: Add openSPOT4

nonoo, 2024-07-12:
Please add the openSPOT4 to tocall.yaml:

- tocall: APOSE?
  vendor: SharkRF
  model: openSPOT4
  class: gadget
  os: Embedded
  contact: [info@sharkrf.com](mailto:info@sharkrf.com)

hessu, 2024-09-29:
Hi,

Looking at the previous use of APOSB? and APOSW? allocations, I see that the wildcard has not been utilised at all. How about using APOSB4 for openSPOT4 and marking it as such in the database? Feels weird allocating complete new prefixes for new versions but not using the wildcard (?) for anything.

nonoo, 2024-09-29:
Yes, we don't use wildcards, those can be omitted. OSW is our product code for the openSPOT2, OSB is for the openSPOT3 and OSE is the openSPOT4. We can release a firmware update for the OS4 to use a different APRS device ID in case APOSE can't be added.

hessu, 2024-09-30:
Seems like quite a lot of APOSE devices are out there in the field already. Do you have automatic firmware updates, do you get a large percentage of devices updated in the end?

nonoo, 2024-09-30:
Yes, automatic upgrade is available, most users let their OS4s update.

hessu, 2024-09-30:
Excellent. How about if I adjust it like this:

```
# prefix of APOSB? for SharkRF (future devices can be added by varying the last character)
 - tocall: APOSB?
   vendor: SharkRF
   contact: info@sharkrf.com

# specific allocations for openSPOT3 and openSPOT4
 - tocall: APOSB
   vendor: SharkRF
   model: openSPOT3
   class: gadget
   os: embedded
   contact: info@sharkrf.com

 - tocall: APOSB4
   vendor: SharkRF
   model: openSPOT4
   class: gadget
   os: embedded
   contact: info@sharkrf.com
```

openSPOT5 can be APOSB5, and so on, in the future.

nonoo, 2024-09-30:
Can we use APSRF? for SharkRF devices?
APOSB4 for the OS4 is fine.

hessu, 2024-09-30:
Unfortunately there is another existing allocation for `APSRF?` already.

closed 2024-09-30 by hessu

nonoo, 2024-09-30:
Ok, APOSB? will be fine.

commit b58c99bd (2024-09-30): + APOSB4 for SharkRF openSPOT4

closes #161
PR #172 (2024-09-30): + APOSB4 for SharkRF openSPOT4
closes #161
commit b4e9211c (2024-09-30): + APOSB4 for SharkRF openSPOT4

closes #161