aprsorg/aprs-deviceid #165: new TOCALL Request: LoRa APRS Balloon software

SQ2CPA, 2024-08-11:
Hello!

I would want to request the TOCALL "APLSKY" for my LoRa APRS balloon software. 

```
tocall: APLSKY
vendor: Damian, SQ2CPA
model: ESP32 LoRa Balloon
class: tracker
os: embedded
contact: sq2cpa@gmail.com

```

That's my first own device ID request so I hope everything is correct :)

Thanks, 73!
Damian SQ2CPA

hessu, 2024-09-29:
Hi,

Unfortunately that would overlap with an existing allocation of 'APLS*' (which is admittedly unnecessarily large, only APLS01 has been used lately, but it is what it is).

How about APLDKY ?

SQ2CPA, 2024-09-30:
Ok, sorry I missed that. 

Please use `APLFLY`, I checked it with '*' and this seems to be free.

hessu, 2024-09-30:
Ok, thanks! Registering as APLFLY.

commit b4a408a8 (2024-09-30): + APLFLY for SQ2CPA ESP32 LoRa Balloon

closes #165
closed 2024-09-30 by hessu