aprsorg/aprs-deviceid #168: APRS TOCALL for BTech UVPRO HT

Chuck-K0XM, 2024-09-26:
tocall: APBTUV
   vendor: BTECH
   model: UV-PRO
   class: HT
   os: 
   contact: support@baofengtech.com
   

Submitted by Chuck Kraly K0XM
k0xm@k0xm.net

hessu, 2024-09-29:
Hi,

Just to confirm: What is your affiliation with BTECH?

I see the UV-PRO being sold already, is it released with firmware using APBTUV or something else?

Chuck-K0XM, 2024-09-29:
I have been developing this radio since we started designs to sell it. I have been assisting in new product development for over 2 years now and also doing customer support, but stay in the background pretty much after my bad history with Bridgecom. 
Yes it is being sold, but we are working on using APBTUV once it is ok, at this time there is no TOCALL in the APRS settings, and adding APRS features like smart beaconing, and messaging (keyboard to keyboard) if possible. The memory limits on this model is restricting us some.
You can inquire to ***@***.*** ***@***.***>  to verify my roles.
 
Chuck K0XM

hessu, 2024-09-30:
Ok! Thanks.

Since there is also an existing APBT62 allocation for BTECH, and no other allocations with the APBT prefix, I'm allocating the prefix APBT* for BTECH and APBTUV for UV-PRO specifically as follows:

```
 - tocall: APBT*
   vendor: BTECH
   contact: support@baofengtech.com

 - tocall: APBT62
   vendor: BTECH
   model: DMR 6x2
   class: ht
   contact: support@baofengtech.com

 - tocall: APBTUV
   vendor: BTECH
   model: UV-PRO
   class: ht
   contact: support@baofengtech.com
```

This allows easy registration of future  BTECH models within that prefix.

closed 2024-09-30 by hessu

hessu, 2024-10-01:
Chuck, one more thing: I've been notified that the UV-PRO supports an iOS app, so it must have BLE (Bluetooth Low Energy) hardware as well.

Does it already support the BLE KISS API: https://github.com/hessu/aprs-specs/blob/master/BLE-KISS-API.md ?

If it supported the BLE KISS API to access the AX.25 TNC functionality, it would immediately work with the following iOS apps:

* aprs.fi APRS application (iPhone & iPad)
* RadioMail Winlink email app (iPhone & iPad)
* PocketPacket APRS app (iPhone & iPad)

It would also be immediately supported by any app which builds support for these other devices which already implement the BLE KISS API:

* Mobilinkd TNC3
* Mobilinkd TNC4
* PicoAPRS v4 transceiver
* RPC Electronics ESP32-APRS Tracker
* Richonguzman / CA2RXU LoRa APRS Tracker on TGO T-Beam and LoRa32v2.1 boards
* [B.B. Link Adapter](https://getbblink.com/) for Kenwood TH-D74/5 radios

This would be a very nice feature for the radio.

Chuck-K0XM, 2024-10-01:
We are getting to that point of development, Please contact me directly at k0xm@k0xm.net as I have a item to discuss along this line.

commit 65ab70f1 (2024-09-30): + APBT* prefix for BTECH, APBTUV for UV-PRO

closes #168