aprsorg/aprs-deviceid #176: request new APRS device id

hb9pae, 2024-10-14:
Dr OM
I'm interested in registering a new APRS divice-id:

tocall: AHB9AK
vendor: SWISS-ARTG
model: LoRa iGate RPI
class: Network
os: Linux/Unix
contact: hb9pae@gmail.com

73 de peter, hb9pae
hb9pae@swiss-artg.ch

hessu, 2024-10-21:
Hi,

Do you have a link to your LoRa firmware branch?

hb9pae, 2024-10-24:
Hi Heikki

yes here it is: 
 
Project discription https://www.swiss-artg.ch/index.php?id=174
Download Sources https://cloud.hoststar.ch/s/GytsTsQSYSoyAXk

Thank you 

73 de peter, hb9pae

hessu, 2024-12-26:
Hi,

Sorry for my slow reply.

The APRS spec (section 4) indicates all software identifiers start with "AP", and the LoRa devices have been going to the APL prefix lately, so I've allocated **APLHB9** for you, as follows. I hope that's ok.

```
 - tocall: APLHB9
   vendor: SWISS-ARTG
   model: LoRa iGate RPI
   class: igate
   os: Linux/Unix
   contact: hb9pae@gmail.com
```

73!

commit 04c3c6f1 (2024-12-26): + APLHB9: SWISS-ARTG LoRa iGate RPI

closes #176
closed 2024-12-26 by hessu