aprsorg/aprs-deviceid #181: Request for new device identifiers

waterflea, 2024-11-08:
Hi,

I would like to request an identifier for our new iGate device. If possible can the tocall start with CSN so we can allocate any future devices under a similar tocall?

tocall: CSN???
vendor: CSN Technologies Inc
model: iGateMini
class: stand alone device
os: Web based
contact: info@igatemini.com

73
Keith
AC2MI
CSN Technologies Inc

hessu, 2024-12-26:
Hi,

Sorry for my slow response.

No, the APRS specification (section 4) says APRS destination callsigns identifying a software starts with "AP". How about "**APSN??**" ?

Since you're developing iGate software, I'd like to use this opportunity pitch the iGate implementation hints document: https://github.com/hessu/aprsc/blob/main/doc/IGATE-HINTS.md

waterflea, 2024-12-31:
No worries hessu, thanks for your time.

We are good with your suggested ID - "APSN??". 

I have read the IGATE-HINTS document a while back during development. I will re-read it just to be sure we didn't miss anything. We are very careful not to gate from the internet to RF unless everything is _just_ right. 

Here are the details for our first device - APSN01:

The class field is difficult since it does basically everything but if we had to pick only one I guess it would be igate.

vendor: CSN Technologies Inc.
model: iGateMini
contact: info@igatemini.com
class: tracker, network, digi, igate
os: embedded
features: messaging

Thank you. Let us know if there is anything you need from us.

Keith
AC2MI
CSN Technologies Inc

closed 2025-03-01 by hessu

waterflea, 2025-03-03:
> completed via 14e4c6b

Hi again. Looks like APSN01 is not going to work. We tried it and it's
coming up as *Brent Hildebrand, KH2Z: APRS+SA (software) *on the aprs.fi
site.

I checked the tocalls file and it looks like he has APS??? reserved. Looks
like that conflicts with APSN.

I checked that APNY?? is available in the tocalls. Can you change *APSN01 *to
*APNY01* and *APSN??* to *APNY??* in the tocalls file?

Thanks for your time.

Keith
AC2MI
CSN Technologies Inc.

commit 14e4c6b1 (2025-03-01): + APSN?? APSN01 for CSN Technologies Inc.

closes #181