aprsorg/aprs-deviceid #182: new TOCALL Request: APRS Balloon software

SQ2CPA, 2024-11-19:
Hello!

I would want to request the TOCALL "APLAIR" for my AFSK APRS + LoRa APRS balloon software based on STM32 from RS41 radiosonde.

```
tocall: APLAIR
vendor: Damian, SQ2CPA
model: APRS Balloon
class: tracker
os: embedded
contact: sq2cpa@gmail.com

```

I already have similar TOCALL "APLFLY" but that's for specific device based on ESP32. Here I want dual modulations AFSK APRS and LoRa APRS so I don't want to use "APLFLY" because that will be missleading.

Thanks, 73!
Damian SQ2CPA

hessu, 2024-12-26:
Hello,

I checked from my database (aprs.fi) and there are 0 stations which have used APLFLY. Could you perhaps use it for this one?

Basically, it makes sense to add a device identifier to this index if there are many stations using that software or device. If it's just a single individual unit or user using that software, it can be described in the comment field, and an experimental APZ* tocall can be used during development. If the software/device is going to be published and distributed, an allocation is useful.

SQ2CPA, 2024-12-27:
Hello,

As I said, I don't want to use the same device id for different devices and APLFLY will be used again for balloons in the summer season because for now ESP32 needs too much power. 

For APLFLY there is a software here: https://github.com/SQ2CPA/LoRa_APRS_Balloon
and for APLAIR also there is a software here: https://github.com/SQ2CPA/RS41_APRS that I already almost tested and will be finished. 

That's why I want another device id and please take into account that I want only one and not the whole like APLA** and then not use it at all...

hessu, 2025-03-01:
Hi,

I've adjusted the existing allocation to be a prefix, so you can do different variations of balloon trackers, just adjust the last letter as necessary (APLFLY, APLFLA, APLFLG...). Further technical details can be easily added in the comment field of the packet, or with a status packet every now and then, they don't necessarily need to be synchronised through the tocalls database.

```
 - tocall: APLFL?
   vendor: Damian, SQ2CPA
   model: LoRa/APRS Balloon
   class: tracker
   os: embedded
   contact: sq2cpa@gmail.com
```

I hope this is alright.

commit 1d9c1fa8 (2025-03-01): + APLFL? for Damian, SQ2CPA

closes #182
commit d9b137d1 (2025-03-01): + APLFL? for Damian, SQ2CPA

closes #182
PR #197 (2025-03-01): + APLFL? for Damian, SQ2CPA
closes #182
closed 2025-03-01 by hessu