aprsorg/aprs-deviceid #184: Request for APUNOn

mzakiab, 2024-12-02:
```
tocall: APUNOn
vendor: 9W2KEY
model: APRSUNO
class: tracker
os: embedded
contact: mzakiab@gmail.com
```

hessu, 2024-12-26:
Hello,

You already have an allocation for APKEYn, where only 1 device is using that allocation. It is not an effective use of the device index if each user gets an allocation for each of their own devices, the index will grow too fast and become hard to maintain. The index is intended for devices and software which are used by many people, and to identify the software, not the user.

Are you planning to distribute this software?

mzakiab, 2024-12-28:
yes, I plan to distribute this software, and in the future will develop more software and other hardware. Thanks

hessu, 2025-03-01:
Hi! Please use either the existing APKEY allocation (which allows numeric versions as the last character), or an experimental tocall in the APZ prefix (APZUNO for example) until the software is ready for distribution. I prefer not to allocate a lot of different tocalls to a single user for software which is not distributed, the database would grow a bit too much if we did that.

closed 2025-03-01 by hessu