aprsorg/aprs-deviceid #186: Mic-E APRSdroid Identifier

na7q, 2024-12-14:
tocall: [1
vendor: Open Source
model: APRSdroid (NA7Q)
class: Software
os: Android
contact: mike.ph4@gmail.com

This is for a custom build of APRSdroid where I have implemented Mic-E encoding with the [1 payload tocall.

hessu, 2024-12-26:
Hi,

I wouldn't like to maintain device identifiers for developer builds and beta forks of software. Especially with mic-e identifiers there aren't that many manufacturer codes available. Are you planning to maintain a fork, or is there an intention to merge this in aprsdroid mainline?

na7q, 2024-12-27:
I am both planning to maintain this fork, and attempting to mainline it as well. I need to discuss with ge0rg about it. My last discussion with him was that he has been too busy, amongst other problems to be able to do any work on APRSdroid. The apps code has remained mostly dormant on updates for quite some time now, until I made some pretty major improvements to it in the last 2 months.

I have the goal of making a Play Store release as well under the same name essentially. Aside from that I publicize it often on Facebook and other social media sites.

Happy 15th birthday to APRSdroid!

ge0rg, 2024-12-27:
I'm going to review the proposed changes to APRSdroid and I would like to mainline them. Even if that doesn't happen on short notice, I'm fully okay with sharing the Mic-E identifier with the APRSdroid forks. Thanks for your work @na7q!

na7q, 2024-12-27:
> I'm going to review the proposed changes to APRSdroid and I would like to mainline them. Even if that doesn't happen on short notice, I'm fully okay with sharing the Mic-E identifier with the APRSdroid forks. Thanks for your work @na7q!

Thanks! I'm hoping to get some PR's going before too long. There are various things I'd like to clean up before then, and some other things I'd like to try to finalize as well.

hessu, 2025-03-01:
Hi!

Sorry for the delay. I'm allocating as follows, leaving contact and author information away as this will be used by the mainline code eventually.

```
 - suffix: "[1"
   vendor: Open Source
   model: APRSdroid
   class: software
   os: Android
```

closed 2025-03-01 by hessu

na7q, 2025-03-01:
Thanks!! Been waiting for this.

commit c0db7914 (2025-03-01): + mic-e identifier [1 for APRSdroid

closes #186