aprsorg/aprs-deviceid #188: Request for APW2Wn

dg3fbl, 2025-01-03:
tocall: APW2Wx
vendor: Joachim Sonnabend, DG3FBL
model: WiresX2Web Software
class: service
os: Windows
contact: mail@dg3fbl.de

HNY @all,
I would like to apply for an APRS identifier for my software. The software runs parallel to the Yaesu Wires-X software on a Windows PC and transmits received positions to the operator's website and now into the APRS network.

best regards, vy 73
Jochen DG3FBL

dg3fbl, 2025-02-25:
any news on this request ?

hessu, 2025-03-01:
Hi!

Sorry for the delay. Allocating as follows:

```
 - tocall: APW2W?
   vendor: Joachim Sonnabend, DG3FBL
   model: WiresX2Web Software
   class: software
   os: Windows
   contact: mail@dg3fbl.de
```

The "service" class is for services such as aprs.fi, and messaging bots, which someone runs for other people as a service, not a "windows service", so I used "software" class instead.

commit 9be04b5d (2025-03-01): + APW2W? for Joachim Sonnabend, DG3FBL: WiresX2Web

closes #188
closed 2025-03-01 by hessu