aprsorg/aprs-deviceid #189: Request an identifier for my new LoRa Tracker and LoRa iGate.

tosunercan, 2025-01-17:
Hello

I would like to request an identifier for my new LoRa Tracker and LoRa iGate devices. Thanks.

- tocall: ?????
   vendor: Ercan, TA3OER
   model: TROY LoRa Tracker
   class: tracker
   os: embedded

- tocall: ?????
   vendor: Ercan, TA3OER
   model: TROY LoRa Igate
   class: Igate
   os: embedded

TA3OER, 73
Ercan TOSUN
KN30ED

hessu, 2025-03-01:
Hello,

Since there have been a lot of duplicate reservations for the same LoRa software previously, please provide a link to your software, or documentation on it.

The intention of the tocalls index is to identify actual different and unique software and firmware. The intention is not to identify the person running a copy of someone else's software.

tosunercan, 2025-03-01:
Hello dear hessu. 

esp32 and Lora software belong to me. Card designs were made by me. 
TA3OER, 73

[https://github.com/tosunercan/TA3OER](url)

[image]
[image]
[Image](https://github.com/user-attachments/assets/1e25ac7e-df59-4043-af48-05702c668e47)

hessu, 2025-04-19:
Hi!

On APRS, each station can use the symbol to signal whether it is a digipeater/gateway, or another sort of station. It is also implicitly visible in the data - igates and digipeaters insert their callsigns in the APRS packet paths, so everyone sees they are digipaters.

Hence, a single software which does both digipeating, igating, and acts as a tracker, does not require multiple tocalls. There are a lot of software packages which do all of these, and it would be weird if they'd have a lot of different destination callsigns assigned to them, when they are in fact a single software doing many things.

Both APLTE? and APLGE? would conflict with other existing allocations, so I made this allocation for you:

```
 - tocall: APLER?
   vendor: Ercan, TA3OER
   model: TROY LoRa Tracker/iGate
   os: embedded
   contact: ta3oer@gmail.com
```

73!

commit faaf5d85 (2025-04-19): + APLER? for Ercan TA3OER LoRa Tracker/iGate

closes #189
commit a90981b4 (2025-04-19): + APLER? for Ercan TA3OER LoRa Tracker/iGate

closes #189
closed 2025-04-19 by hessu