aprsorg/aprs-deviceid #191: Request an identifier for my new LoRa Tracker and LoRa iGate.

ikram8484, 2025-01-27:
Hello

I would like to request an identifier for my new LoRa Tracker and LoRa iGate devices. Thanks.

tocall: APTGIK
vendor: Juliet Delta, 9M4GIK
model: LoRa Tracker 
class: Tracker
os: embedded
contact: [9m2ikr@gmail.com]

tocall: APGGIK
vendor: Juliet Delta, 9M4GIK
model: LoRa Igate
class: iGate
os: embedded
contact: [9m2ikr@gmail.com]

9M2IKR, 73
Juliet Delta

ikram8484, 2025-02-26:
Hai ang update my request.

hessu, 2025-03-01:
Hello,

Since there have been a lot of duplicate reservations for the same LoRa software previously, please provide a link to your software, or documentation on it.

The intention of the tocalls index is to identify actual different and unique software and firmware. The intention is not to identify the person running a copy of someone else's software.

ikram8484, 2025-03-02:
I will change to Other software Not Lora Device. but my lora still under my build cannot share after finish. tq.

ikram8484, 2025-03-19:
Anyupdate for this request

ikram8484, 2025-03-21:
I change the call id because my lora not finish yet..but i change to normal
lora firmware not lora firmware

hessu, 2025-04-19:
Hello,

Sorry for the delay.

On APRS, each station can use the symbol to signal whether it is a digipeater/gateway, or another sort of station. It is also implicitly visible in the data - igates and digipeaters insert their callsigns in the APRS packet paths, so everyone sees they are digipaters and igates.

Hence, a single software which does both digipeating, igating, and acts as a tracker, does not require multiple tocalls. There are a lot of software packages which do all of these, and it would be weird if they'd have a lot of different destination callsigns assigned to them, when they are in fact a single software doing many different things.

I believe this request is for a single software which can act both as a tracker and gateway, hence I am allocation only a single identifier for now. This reduces the growth rate of the allocation file.

```
 - tocall: APTGIK
   model: APRS Melaka
   os: embedded
   contact: 9m2ikr@gmail.com
```

If you need a different identifier for a **different software** on LoRa, or if you have additional changes or needs. please open a new ticket. Thanks!

73!

commit 19db98c1 (2025-04-19): + APTGIK for 9M4GIK APRS Melaka

closes #191
closed 2025-04-19 by hessu