aprsorg/aprs-deviceid #192: Requesting a TOCALL for aprs-weather-submit

rhymeswithmogul, 2025-02-14:
Hello, everyone!  I'm the author of [`aprs-weather-submit`](https://github.com/rhymeswithmogul/aprs-weather-submit).  This free and open-source app allows amateurs to submit weather data at will through a command-line interface, or it can be called automatically by a script or other daemon as a building block to create your own APRS-IS-capable weather station!

I'd like to register a TOCALL for it:

```yaml
- tocall: APRSWX
  vendor: Colin Cogle, W1DNS
  model: aprs-weather-submit
  class: software
  os: Multiple
  contact: https://github.com/rhymeswithmogul/aprs-weather-submit/
```

If the `APRS??` prefix is reserved, I don't mind thinking of something else.

Also, does the contact field _need_ to be an email address?  I'd prefer it link to my GitHub project page where I can collect and triage bug reports and discussions.  If not, I have no problem submitting my email address.

hessu, 2025-03-01:
Hi!

There is no formal reservation for APRS?? prefix, but the APRS tocall itself is the reserved default tocall. Let's maybe leave that prefix alone for now, just in case.

How about an allocation of APWXS? prefix, leaving one character for versioning?

I suppose the contact address can be a github link just as well.

rhymeswithmogul, 2025-03-01:
Sure, APWXS? works for me. I don't think I need the versioning digit, but you never know what might happen.

hessu, 2025-04-19:
Hi,

Sorry for the delay. Allocated APWXS? now.

Adjusted class to "daemon", while this is not really a long-term-running daemon at the moment, the definition for "software" is "Desktop software", while "daemon" stands for "Computer software without user interface" which fits better.

commit be9646fe (2025-04-19): + APWXS? for W1DNS aprs-weather-submit

closes #192
closed 2025-04-19 by hessu