aprsorg/aprs-deviceid #198: APFIRE

na7q, 2025-03-07:
tocall: APFIRE
vendor: Mike, NA7Q
model: National Interagency Fire Center Alerts
class: Service

contact: [mike.ph4@gmail.com](mailto:mike.ph4@gmail.com)

hessu, 2025-04-19:
Hi,

This would conflict with the existing allocation for aprs.fi (APFI??). How about APNIFC ?

The device identifiers are  for APRS software and hardware. "National Interagency Fire Center Alerts" sounds like a source of data, not APRS software. What is the software used to forward this traffic?

hessu, 2025-09-03:
Hi,

Pinging again on this.

This would conflict with the existing allocation for aprs.fi (APFI??). How about APNIFC ?

hessu, 2025-09-29:
Thanks! Allocating as follows:

```
 - tocall: APNIFC
   vendor: Mike, NA7Q
   model: National Interagency Fire Center Alerts
   class: service
```

73!

commit 147a2ba2 (2025-09-29): + APNIFC for NA7Q, National Interagency Fire Center Alerts

closes #198
closed 2025-09-29 by hessu