aprsorg/aprs-deviceid #200: Add APNV02

sq8vps, 2025-03-10:
Hi,
please add APNV02:
```
   tocall: APNV02
   vendor: SQ8L
   model: VP-Digi on AIOC
   os: embedded
   class: digi
```
Thanks!

hessu, 2025-04-19:
Hi,
This, I believe, is still the same VP-Digi software, just running on AIOC hardware?

sq8vps, 2025-04-20:
Yes, but this is a different hardware platform and a different firmware build. Just wanted to differentiate on aprs.fi.

hessu, 2025-04-20:
Ok, I understand.

There is a lot of different cross-platform software with different build variants out there. If we included all different builds, it would be a lot of extra work for me to update the information for all of them.

Since APNV0? is allocated for VP-Digi, you can use APNV02 for the variant, but I would prefer not to maintain such small build differences here manually.

sq8vps, 2025-04-20:
Ok, that seems fair. Thank you!

closed 2025-04-20 by sq8vps