aprsorg/aprs-deviceid #203: Request ADD TNC AEA PK87 / PK88 / PK232

F4GXS, 2025-04-05:
Good evening !

Please, can you add those "tocall" for 3 TNC from **AEA** _(Advanced Electronic Applications)_ **PK-87**, **PK-88** and **PK-232**.

Working in digipeater mode, all tested OK and perfectly working autonomously.

Those "tocall" will be added in the "**UNPROTO**" parameter from TNC Command.

```
 - tocall: APAE87
   vendor: AEA
   model: PK-87 TNC

 - tocall: APAE88
   vendor: AEA
   model: PK-88 TNC

 - tocall: APAE23
   vendor: AEA
   model: PK-232 TNC
```

Thank you !

F4GXS.

hessu, 2025-04-19:
Hi,
I'm not sure if these count as APRS hardware. What APRS features do they have, do they even do New-N digipeater paths?

F4GXS, 2025-04-19:
Hi,
those hardware are not connected to APRS-IS or Internet or something, it's only digipeating packets, kind of interesting if you have handheld low power device and the "hardware TNC" are in top of moutain, increasing by a lot the distance range of the APRS Packet transmited by the handheld, allready tested with my Yaesu VX-8 in Low Power Mode "L1" at 50mW in middle of nowhere, i can do ~+50Km
(~+31 Miles) depending on the power of the TNC digipeater radio.

What do you mean about New-N ?

For the digipeater paths, it can be choose, into the setup of the TNC, an exemple : for the setup of a PK-232, in the "Unproto" parameter it wil be "APAE23 V WIDE1-1, WIDE2-1"
For those "WIDEn-n" it's all custom, i can chose whatever a want but only for the resonable, i don't want to exagerated.
I sayd "APAE23 V WIDE1-1, WIDE2-1" but it can be other setup as needed

The only problem that I encountered and which is not very serious for me is that the "WIDE1-1" is not changed by the hardware call, like a Call Fill-in.

Exemple : Call of the TNC is "N1FILL"
EX4MPL>APEA23,WIDE1-1,WIDE2-1:data
EX4MPL>APEA23,N1FILL,WIDE2-1:data

The WIDE1-1 is not changed by the call, there are no fill-in

In that case, if the "Unproto" is set to "APEA23 V WIDE1-1, WIDE2-1", digipeated packet will be showed like this : EX4MPL>APEA23,WIDE1-1*,WIDE2-1:data
Which is totally fine, except you just have to think a little to know where the frames are digipeated.

I'm on the way to test other TNC in time, other TNC Hardware are comming very soon for other test.

hessu, 2025-04-19:
Hi,

Until now the APRS device identifiers have only been allocated to **APRS devices**, which implement actual APRS digipeating (or other parts of the APRS protocol).

Generic TNCs without APRS path functionality cannot decrement the SSIDs in the path element (WIDE2-2 -> WIDE2-1, WIDE2-1 -> WIDE2*). They do not implement APRS digipeating and with some misconfiguration will retransmit the same packet many times. They are not good for the purpose. There are good APRS digipeater software out there which can use these generic KISS TNCs for modulation/demodulation, and there are also some standalone TNCs which do implement the APRS path logic.

So, for now, I won't be adding all the generic TNCs here. In a pinch you can use them as a WIDE1-1 fill-in digipeater at home if you don't have anything better, just use the generic "APRS" tocall or pick an experimental one in the "APZ*" range. Putting these on a mountaintop as a "WIDE2-2" digipeater will be problematic as they will retransmit the same packet multiple times when they receive it from more than one other digipeater.

https://blog.aprs.fi/2020/02/how-aprs-paths-work.html

closed 2025-04-19 by hessu

F4GXS, 2025-04-19:
Well, thanks...